GiorgioAresu/FanController #1: How to maintain fan(4 pins) speed?

Iflier, 2017-08-20:
Hello, great work! But I have a question: How do I keep the fan at a fixed speed? En, the fan speed can be read normally, when i send a string to arduino through serial, the fan can not fixed at a  special speed. When the serial port accepts input, the speed changes, and then back to the previous speed. What should I do?

GiorgioAresu, 2017-08-20:
Hi, by how much does the speed change? Can you hear the change in the fan sound or the difference is just reported by the Arduino?

Iflier, 2017-08-25:
Yes, I can.  Only when I send string to Arduino, fan speed is changed, but after a while, it seems that nothing has happened.

GiorgioAresu, 2017-08-28:
I'll take a look next week and report back

GiorgioAresu, 2017-09-09:
Can you be more specific on the steps your doing please? Either I can't reproduce the problem or I didn't get exactly what you meant because I left it going for a couple of hours and speed remained stable. Thanks

GiorgioAresu, 2018-05-07:
Closing due to inactivity

closed 2018-05-07 by GiorgioAresu