GiorgioAresu/FanController #2: don't work on nodemcu

fireport, 2017-11-22:
I'm trying to use your library on a nodeMCU but the RPM counter is always 0.
The only change that i did in your example code is from #define SENSOR_PIN 2 to #define SENSOR_PIN D5. 
The code attached that use the same pin and the plain vanilla interrupt library works without problem
`const byte interruptPin = D5;
volatile byte interruptCounter = 0;
int numberOfInterrupts = 0;
 
void setup() {
 
  Serial.begin(115200);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), handleInterrupt, FALLING);
 
}
 
void handleInterrupt() {
  interruptCounter++;
}
 
void loop() {
 
  if(interruptCounter>0){
 
      interruptCounter--;
      numberOfInterrupts++;
 
      Serial.print("An interrupt has occurred. Total: ");
      Serial.println(numberOfInterrupts);
  }
 
}`
Do you have any suggestions ?

fireport, 2017-11-22:
Using pinMode(SENSOR_PIN , INPUT_PULLUP) your library start working but the nodemcu reset after a few read

GiorgioAresu, 2017-11-22:
Hi, I'll give it a look when I have a little time, probably this weekend. If you get to it before me you could try messing with the library source, I can't tell the problem right off the bat.

photodude, 2019-03-20:
@fireport if the duty cycle is resetting after a few reads then you might have the wrong serial monitor setting. I experienced this resetting, I would enter a value in the serial monitor, hit enter the value would change then would go to zero. This was due to my serial monitor setting being set to "newline" the serial would see the int value and set the duty cycle I wanted then it would see the new line character and `serial.parseInt()` would convert the new line to a zero. 

Documentation regarding setting the serial monitor to "no line ending" is lacking.

GiorgioAresu, 2019-07-13:
Sorry for the time passed. I'm closing this, if you still have problems feel free to reopen it or open a new one

closed 2019-07-13 by GiorgioAresu