GiorgioAresu/FanController #3: Could I change the pwm frequency?

glauberbiro, 2018-05-31:
Hi. thanks for your library.
Could I change the pwm frequency? cuz my fan is very noisy. thank you.

GiorgioAresu, 2018-06-01:
Hi,
it is possible, but it may break other stuff since it breaks timing functions. In fact it would probably break even the rpm counter of the library itself. I didn't try it because I don't have an oscilloscope and my fans are not noisy at the default frequency, even if it's off-spec. 

Here you can find more info:
- [PwmFrequency](https://playground.arduino.cc/Code/PwmFrequency)
- [SecretsOfArduinoPWM](https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM)

If you try it please let me know how it goes, if the fans become quieter and if the speed still work. What brand/model are your fans?

glauberbiro, 2018-06-01:
Hi Giorgio!

Thanks for your contact.

ok, I see.

I have seen this topics about the arduino pwm as well, but I'm using the NODEMCU ESP8266-12E. Today I found some topics about the ESP PWM and the timers and pins are diferent from arduino like this under. I´m work now and later I'll test another things at home.

https://smartarduino.gitbooks.io/development-of-nodemcu/content/subsection_43_pwm_function.html

I´m using ADDA and intel fans, anex pictures. The intel fans is recommended to use 25kHz.

Glauber Trevisol Furini

[image]

GiorgioAresu, 2018-06-01:
Seems like that too is limited to a lower frequency than what you need. If you really need a frequency that high, meaning it won't work correctly with a way lower one, maybe you're better off with something like an [SG3525](https://www.openimpulse.com/blog/wp-content/uploads/wpsc/downloadables/SG3525A-Datasheet.pdf) or similar.

[Apparently](https://arduino.stackexchange.com/questions/4610/how-to-eliminate-noise-from-pwm-controlled-12v-fan-at-low-speed) someone managed to reduce the noise with the opposite approach, maybe it works for you too.

glauberbiro, 2018-06-01:
Hi Giorgio!

I saw that topic yesterday, btw thanks for sharing.

I arrived at home and changed the pwm frequency to a lower value, from
default 1000 Hz to 60 Hz. and the sound is better now, but usually the PWM
freq to drive motor is higher them this, look the intel fan specs "PWM
Frequency: Target frequency 25 kHz, acceptable operational range 21 kHz to
28 kHz".

I read the topic and tested a serie capacitor, as well, and works. But the
PWM essencial is modulating, when you use a serie capacitor the pulse turns
linear and all the PWM purpouse does't exist anymore, like efficient energy
and pid controls). I could do something like that, like use a LR/LC/LRC
filters, but I don't know how to calculating and I belive in the correct
frequency will works well.

The fan cooler from the topic, that you send to me, is not built to work
with PWM control, but the 4 wire is built for it. Take a look at the intel
4 wire fan specs.

Thanks for your help.

PS: I'm doing this project to my keep cool my italian wines

GiorgioAresu, 2018-06-02:
Sorry, I didn't read carefully and skipped the part saying it was a 3 wire fan.
I know about the PWM frequency spec being so high, it's just that the only fans I tried with were 2 Noctua ones, and they do not seem to complain for it.

I'm a bit concerned now because the whole point of this library was to add fans to my rack, since during summer I have to keep its door open. I'd consider cheaper fans, but it seems that comes at the risk of them being noisy with out of spec freqs. What kind of noise do they produce? Like a whine or something?

Cool (get it? :smile: ), but can I ask, what purpose would they serve to cool something that's not actively dumping heat in the room? Do you have some other type of cooling maybe?

glauberbiro, 2018-06-11:
> Cool (get it? 😄 ), but can I ask, what purpose would they serve to cool

Hi. Sorry for the late answer.

I found this kind of shields, both of then with serial communication. I
think will resolve my problem, and use your library and the esp8266 only to
read the tach sensor.

https://www.aliexpress.com/item/DC-5-30V-Micro-USB-5V-Power-Independent-PWM-Generator-2-Channel-Dual-Way-Digital-LED/32839361904.html?spm=a2g0s.9042311.0.0.41714c4dVHe1Db

https://www.aliexpress.com/item/Signal-Generator-PWM-Pulse-Frequency-Duty-Cycle-Adjustable-Module-LCD-Display-1Hz-150Khz-3-3V-30V/32835541668.html?spm=a2g0s.9042311.0.0.41714c4dVHe1Db

Is a typical swiching frequency noise, from the inductive load l. Try to
see you see this video, please. https://photos.app.goo.gl/cFxg9WuLvz9WANdv8

Yes, its a Thermoelectric Peltier Wine Cooler. I can use any frequency,
doesn't make any noise, cause is a resistence load.

thanks for your help.

GiorgioAresu, 2018-06-13:
Nice, good to know they're pretty cheap too, they might come handy in the future.
Gosh, that noise 🤣
Oh I see.

Anytime

closed 2018-06-13 by GiorgioAresu