GiorgioAresu/FanController #6: How to control multiple fans?

photodude, 2019-03-20:
The library readme says "supports up to 6 fans" but there is no documentation or examples in the library for controlling up to 6 fans.

How do I control 6 fans with this library?

GiorgioAresu, 2019-05-04:
Hi, if you take a look at the Arduino [docs about interrupts](https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/) you can see where the _up to 6_ comes from. This depends on the model you are using, and it seems that some models even go past 6, though I have not tried it. In the library examples you can see how the sensor pin is passed, just by the pin number as per the same documentation

photodude, 2019-05-05:
I see, so this is written to control a group of fans at the same speed with up to 6 pins on interrupts for reading the sensor feedback on each fan.

Looks like that's not what I'm looking for. I was looking for individual control over 6 individual fans. ( I can do it, with calling the library multiple times). guess it doesn't really matter as PC 4-pin PWM fans were removed due to the limitations imposed by the 4-pin PC fan Specifications (there is no "stop" command, and the lowest speed is 30% PWM signal)

Thank you.

closed 2019-05-05 by photodude

GiorgioAresu, 2019-05-05:
Please take a look at the _MonitorAndControl_ example:
```c
// Initialize library
FanController fan(SENSOR_PIN, SENSOR_THRESHOLD, PWM_PIN);
```
For each fan you specify both a sensor pin and a pwm pin, so you can control each one individually. I don't have an Arduino mega so I cannot guarantee (I doubt that it will remain accurate) on how it will give accurate readings with that many interrupts firing.
Yes, the protocol is not the best, I ended up using a cheap pwm controller from aliexpress, but I'm going to give it another try with an esp32 when I have some time

photodude, 2019-05-06:
Yes, that is how I initialized the library for each fan

```cpp
// Initialize library
FanController fan0(SENSOR_PIN0, SENSOR_THRESHOLD0, PWM_PIN0);
FanController fan1(SENSOR_PIN1, SENSOR_THRESHOLD1, PWM_PIN1);
FanController fan2(SENSOR_PIN2, SENSOR_THRESHOLD2, PWM_PIN2);
// Repeat as needed for the number of PWM fans being controlled
```
This worked fine on the mega (as long at you are aware of all the PWM timer related issues that can come up). 
I will note that this method likely would cause issues eventually as each instance of the library can take six interrupts for the sensor. When controlling the fans individually there is no need for each instance of the library to have 6 interrupts

If I was going down the road of 4-pin PWM PC fans I would likely write a library using the TimerOne/TimerThree/TimerFour/TimerFive libraries and the fan speed example there as it would give the 25kHz PWM signal that the 4-Pin PWM fans require. https://github.com/PaulStoffregen/TimerOne/blob/master/examples/FanSpeed/FanSpeed.pde
that group of libraries are specifically designed for working with getting the PWM timers right without breaking all of the other timer-based functions like mills() on Timer 0 that Arduino requires.

but as I mentioned, we dropped the 4-pin PWM fans from our design options as the protocol for those fans have too many limitations and thus do not meet our needs. 

Thanks again for having a library that was available. This helped get us started in testing these as options for our project.