GiorgioAresu/FanController #7: Not working with new Arduino Version (ESP32)

DougJunior, 2019-07-06:
Hi,

trying to use it on a new ESP32, but you use in your Scipt stuff like AnalogWrite this is allready outdated and get no more used in the new Arduino Version. Plz update this nice biliothek.

Complett Error Message when i try a startup:
 `Arduino: 1.8.9 (Windows 7), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

WARNUNG: Bibliothek LiquidCrystal_I2C behauptet auf (avr) Architektur(en) ausgeführt werden zu können und ist möglicherweise inkompatibel mit Ihrem derzeitigen Board, welches auf (esp32) Architektur(en) ausgeführt wird.
C:\Users\DougJunior\Documents\Arduino\libraries\FanController\FanController.cpp: In member function 'void FanController::setDutyCycle(byte)':

C:\Users\DougJunior\Documents\Arduino\libraries\FanController\FanController.cpp:40:36: error: no matching function for call to 'min(byte&, int)'

  _pwmDutyCycle = min(dutyCycle, 100);

                                    ^

In file included from c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\algorithm:62:0,

                 from C:\Users\DougJunior\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:148,

                 from C:\Users\DougJunior\Documents\Arduino\libraries\FanController\FanController.cpp:1:

c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:3451:5: note: candidate: template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)

     min(initializer_list<_Tp> __l, _Compare __comp)

     ^

c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:3451:5: note:   template argument deduction/substitution failed:

C:\Users\DougJunior\Documents\Arduino\libraries\FanController\FanController.cpp:40:36: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'

  _pwmDutyCycle = min(dutyCycle, 100);

                                    ^

In file included from c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\algorithm:62:0,

                 from C:\Users\DougJunior\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:148,

                 from C:\Users\DougJunior\Documents\Arduino\libraries\FanController\FanController.cpp:1:

c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:3445:5: note: candidate: template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)

     min(initializer_list<_Tp> __l)

     ^

c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:3445:5: note:   template argument deduction/substitution failed:

C:\Users\DougJunior\Documents\Arduino\libraries\FanController\FanController.cpp:40:36: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'

  _pwmDutyCycle = min(dutyCycle, 100);

                                    ^

In file included from c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\algorithm:61:0,

                 from C:\Users\DougJunior\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:148,

                 from C:\Users\DougJunior\Documents\Arduino\libraries\FanController\FanController.cpp:1:

c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:243:5: note: candidate: template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)

     min(const _Tp& __a, const _Tp& __b, _Compare __comp)

     ^

c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:243:5: note:   template argument deduction/substitution failed:

C:\Users\DougJunior\Documents\Arduino\libraries\FanController\FanController.cpp:40:36: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')

  _pwmDutyCycle = min(dutyCycle, 100);

                                    ^

In file included from c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\algorithm:61:0,

                 from C:\Users\DougJunior\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:148,

                 from C:\Users\DougJunior\Documents\Arduino\libraries\FanController\FanController.cpp:1:

c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:195:5: note: candidate: template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)

     min(const _Tp& __a, const _Tp& __b)

     ^

c:\users\dougjunior\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:195:5: note:   template argument deduction/substitution failed:

C:\Users\DougJunior\Documents\Arduino\libraries\FanController\FanController.cpp:40:36: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')

  _pwmDutyCycle = min(dutyCycle, 100);

                                    ^

C:\Users\DougJunior\Documents\Arduino\libraries\FanController\FanController.cpp:41:43: error: 'analogWrite' was not declared in this scope

  analogWrite(_pwmPin, 2.55 * _pwmDutyCycle);

                                           ^

exit status 1
Fehler beim Kompilieren für das Board DOIT ESP32 DEVKIT V1.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.`

GiorgioAresu, 2019-07-08:
Hi, it should be fixed with v1.0.4. Please see README.md and let me know if it solves the issues

DougJunior, 2019-07-09:
Hi again, thx for the work, now i can compile the code.
But i get a reading from the fan with zero rpm and when i try to adjust the fan speed with somethink like 80 or 50 i get only a high pitch sound from the fan.
I found questerday out that when i get with a 25mhz signal on the fan, he gives the readings for the rpm in the same way back. I tryed to be sure a second fan.

GiorgioAresu, 2019-07-09:
Sorry to hear that. I'll find an ESP32 and I'll take a look. I got myself an oscilloscope now, so it should be easier to trace the issue. The ESP32 should have less problems than the arduino, since it should be able to output frequencies compliant to 4 wire pwm fan specs

DougJunior, 2019-07-10:
I think i found the problem, the ESP32 trys to communicate with the fan with 5 KHz, 
I buyed beforehand a DSO Shell 150, extrem cheap and for such small thinks it does the job.

GiorgioAresu, 2019-07-10:
So that's the problem :) It should be a matter of setting the right parameter to the underlying library then. Nice :)

GiorgioAresu, 2019-07-13:
Hi, it should be ok with v1.0.5, I tried with the oscilloscope and the signal is now at 25kHz

DougJunior, 2019-07-15:
I tested it again, i getting Zero RPM and now i can control the fan for 3 or 4 seconds. After that it goes back to 100%.

I found a other soultion.
`#define anemo 26//Variablen für Anemo
float averageAnemo = 0;
int freq = 25000;
int luefter_channel =15;
int resolution = 8;
int luefter_power = 18;

setup{
  ledcSetup(luefter_channel,freq,resolution);
  ledcAttachPin(luefter_power, luefter_channel);
  ledcWrite(luefter_channel,127);
}`

to control the fan, i use this:  You need mini 15%, under it i got issues withe the fan.
`  if(rotary_counter >2)
  {
    int cycle = rotary_counter*5*2.55;
    Serial.println(cycle);
    ledcWrite(luefter_channel,cycle);
  }`
When the fan is running i use this for the speed monitoring:
`    unsigned long tt1 = pulseInLong(anemo,HIGH);
    delay(50);
    if(tt1<100 ) tt1= tt1*1000;
    unsigned long tt2 = pulseInLong(anemo,HIGH);
    delay(50);
    if(tt2<100) tt2= tt2*1000;
    unsigned long tt3 = pulseInLong(anemo,LOW);
    delay(50);
    if(tt3<100) tt3= tt3*1000;
    unsigned long tt4 = pulseInLong(anemo,LOW);
    if(tt4<100) tt4= tt4*1000;
    unsigned long tt = max(tt1,tt2)+max(tt3,tt4);
    //unsigned long tt = max(pulseIn(anemo,HIGH,40000),pulseIn(anemo,HIGH,40000)) + max(pulseIn(anemo,LOW,40000),pulseIn(anemo,LOW,40000));
    Serial.print(tt1);
    Serial.print("\t");
    Serial.print(tt2);
    Serial.print("\t");
    Serial.print(tt3);
    Serial.print("\t");
    Serial.print(tt4);
    Serial.print("\t");
    Serial.println(tt);
    if (tt > 500)
    {
    averageAnemo = (((1000000/(tt*2))* 60) + (3* averageAnemo))/4;  
    Serial.print("RPM Speed ");
    Serial.println(averageAnemo);`

The most from this is trash code for testing but maybe you can use it.

GiorgioAresu, 2019-07-15:
I'll try again then. That's basically what I'm doing, except for using a wrapper library ([ESP32_AnalogWrite](https://github.com/ERROPiX/ESP32_AnalogWrite)), but setting 10bit resolution and 25KHz ([here](https://github.com/GiorgioAresu/FanController/blob/master/FanController.cpp#L23). Maybe it's the wrapper lib?

DougJunior, 2019-07-15:
Not sure what the wrapper lib does,
i hade the problem with ledcSetup that when i set it up pinmode, that it not worked.
I using arduino real for the first time, and i current working on a controll system for a enclosed 3d printer envirement. So i learn it step by step and build it together.

closed 2020-01-25 by GiorgioAresu