GiorgioAresu/FanController #8: pwm frequency 500Hz

Steph71, 2019-07-29:
Hi,

I am trying this library on my arduino nano, but I can see, that the PWM Frequency is ~500Hz.

But its supposed to be ~ 25kHz right ?

Am I missing something here ?

Best Regards
Stephan

GiorgioAresu, 2019-07-29:
Hi, "classic" arduinos are not powerful (fast) enough to generate high
frequency pwm signals, esp32 based boards are. Still, with arduinos you are
generally going to get close a decent result as long as the fans are not
too picky (I tried noctua fans I had around and they worked just fine), but
we are talking about frequencies way out of specifications, so you cannot
be sure which ones are going to work. If your project needs reliability you
can try an esp32 or going straight to a pwm controller board. Theoretically
you are not supposed to control and monitor pwm fans straight with/from
pins without mosfets and stuff, so take this library as a quick and dirty
hack

Steph71, 2019-07-30:
Hi, thanks for your quick reply. 

actually there are libraries which can generate the 25kHz PWM on an arduino (eventually not 6 in parallel) e.g.:

https://code.google.com/archive/p/arduino-pwm-frequency-library/
or 
https://github.com/marceloaqno/4-Wire-FAN-Arduino

however, those do not include the monitoring code.

and you are right the direct coupling of the controller an the fan output might cause reliability issues.

are you aware of any of the shelf modules, that can provide spec compliant and reliable interfacing between PC fan and generic microcontrollers. 

Actually I would assume, this to be a very generic components for a lot of projects ....

Best Regards
Stephan

GiorgioAresu, 2019-07-31:
Hi, I saw some of those libraries, I always assumed they would mess up
timings due to changing the registries (
https://forum.arduino.cc/index.php/topic,117425.0.html) if you were to use
millis and such. Never actually tried them though.
Unfortunately I didn't find PWM interface modules to connect to
microcontrollers, only standalone ones. Some had ways to communicate with
them via serial interface IIRC, though I didn't buy one of those due to
their size. I ended up with a standalone one, I'm not entirely satisfied,
but it does the job.

Have a nice day :)

closed 2019-12-11 by GiorgioAresu