GiorgioAresu/FanController #10: ISR not in IRAM problem on ESP8266

AndSte01, 2019-10-28:
on newer versions of the ESP8266 ISRs must be placed in the IRAM not in the flash.
This could be done by adding the following lines to your code:
```
void ICACHE_RAM_ATTR FanController::_triggerExt0 ();
void ICACHE_RAM_ATTR FanController::_triggerExt1 ();
void ICACHE_RAM_ATTR FanController::_triggerExt2 ();
void ICACHE_RAM_ATTR FanController::_triggerExt3 ();
void ICACHE_RAM_ATTR FanController::_triggerExt4 ();
void ICACHE_RAM_ATTR FanController::_triggerExt5 ();
```

GiorgioAresu, 2019-12-11:
Hi, thanks for taking the time for opening the issue. Version v1.0.6-rc1 should fix the issue, can you confirm?

AndSte01, 2019-12-11:
The code compiles fin now.
I will test this on a board as soon as it is possible.

AndSte01, 2019-12-14:
I've tested it on hardware as well. Everything works fine.

closed 2019-12-14 by AndSte01