GiorgioAresu/FanController #9: compiler warning due to inappropriate placement of return statement

AndSte01, 2019-10-28:
in `unsigned int FanController::getSpeed()` the return statement is wrapped inside an if statement, making it possible that nothing is returned. As a result you get the "warning: control reaches end of non-void function" compiler warning.

a possible fix would be:
```
unsigned int FanController::getSpeed() {
	unsigned int elapsed = millis() - _lastMillis;
	if (elapsed > _sensorThreshold)
	{
		detachInterrupt(_sensorInterruptPin);
		double correctionFactor = 1000.0 / elapsed;
		_lastReading = correctionFactor * _halfRevs / 2 * 60;
		_halfRevs = 0;
		_lastMillis = millis();
		_attachInterrupt();
	}
	return _lastReading;
}
```
This would return the last Reading even if there isn't a new available

GiorgioAresu, 2019-12-11:
Hi, thanks for taking the time for opening the issue. Version v1.0.6-rc1 should fix the issue, can you confirm?

AndSte01, 2019-12-11:
The compiler warning is now gone.
Due tu my lack of time I can't test it on a real board, but as soon as I can I will respond.

AndSte01, 2019-12-14:
I've tested it on hardware as well. Everything works fine.

closed 2019-12-14 by AndSte01