GiorgioAresu/FanController #11: FanController constructor truncates "sensorThreshold" to 8 bits.

MojaveTom, 2019-11-02:
**FanController.h**
```
class FanController
{
	public:
		FanController(byte sensorPin, unsigned int sensorThreshold, byte pwmPin = 0);
		void begin();
		unsigned int getSpeed();
		void setDutyCycle(byte dutyCycle);
		byte getDutyCycle();
	private:
		static FanController *_instances[6];
		byte _sensorPin;
		byte _sensorInterruptPin;
		byte _sensorThreshold;       //    <--- aka *uint8_t*
		byte _pwmPin;
		byte _pwmDutyCycle;
		byte _instance;
		unsigned int _lastReading;
		volatile unsigned int _halfRevs;
             . . . 
```
**FanController.cpp**
```
FanController::FanController(byte sensorPin, unsigned int sensorThreshold, byte pwmPin)
{
	_sensorPin = sensorPin;
	_sensorInterruptPin = digitalPinToInterrupt(sensorPin);
	_sensorThreshold = sensorThreshold;      //  <---  assign *unsigned int* to *byte*
	_pwmPin = pwmPin;
	pinMode(pwmPin, OUTPUT);
	_pwmDutyCycle = 100;
}
```

GiorgioAresu, 2019-12-11:
Hi, thanks for taking the time for opening the issue. Version v1.0.6-rc1 should fix the issue, can you confirm?

MojaveTom, 2019-12-12:
The code mods appear to fix this issue.  I have run no tests to verify; but the fix is so simple that tests should not be necessary.

closed 2020-01-25 by GiorgioAresu

MojaveTom, 2020-01-25:
Sorry I didn’t respond earlier.  I agree the issue seems to be resolved by looking at the code.  I haven’t had need to use that module recently.

Thanks,
Tom

GiorgioAresu, 2020-01-25:
No problems, I'm not very active either, I just remembered I never promoted
it to stable :) have a nice day