GiorgioAresu/FanController #12: Frequency of PWM @ 25kHz not taken into account?

bobwolff68, 2020-04-14:
https://github.com/GiorgioAresu/FanController/blob/785590adb448aaf86e80e43dbc44183c31855085/FanController.cpp#L55

I love the way you architected this class. Seriously very nice. And I was quite excited to use it with a 4-pin PC fan project I'm currently working on. But when I looked at the code, I found this analogWrite() call for the PWM output. I believe this won't work for a 4-pin fan with PWM control as the PWM signal cannot be the 500Hz frequency PWM native to Arduino but instead must be a 25kHz frequency signal. The examples I've seen (which are NOT done well as you've done your class) utilize the internal timers attached to very specific pins on particular models of Arduinos etc. Can you comment on what you've tested this against or your thoughts on the use of the PWM here?
Thanks!

xavser, 2020-04-15:
Hi!
Thanks a lot for the library, awesome job!!
I just had the same timer frequency issue.
It actually works well if you set them to 31372khz by dividing frequency per 1.
to try it, roughly, you can add this in the setup():
   TCCR2B = TCCR2B & B11111000 | B00000001; //timer 2  31372HZ pin 3 & 11
   TCCR1B = TCCR1B & B11111000 | B00000001; //timer 1  31372HZ pin 9 & 10

> https://github.com/GiorgioAresu/FanController/blob/785590adb448aaf86e80e43dbc44183c31855085/FanController.cpp#L55
>
> I love the way you architected this class. Seriously very nice. And I was quite [...]

GiorgioAresu, 2020-04-15:
Hello, thank you very much :)
Yes, unfortunately I am aware that the native Arduino PWM frequency is way lower than what it should be for PWM fans, but as you already found, changing registries affecting the internal timers is not so well documented, and the fans I had around worked just fine on that frequency, even though I received feedback from other people that weren't so lucky. So I decided to leave it as such, as I was afraid to ruin compatibility with other functions and libraries that may rely on that. I eventually moved away from using it but I'm trying to support it as best as I can, so I'd love to fix this. Do you have any reference or experience on whether changing the registries breaks something?

bobwolff68, 2020-04-16:
Thanks for the explanation. I will definitely need the 25kHz (or if 31 works...) on mine. I'll give a try with Xavser's approach, but the tricky tricky items are: A) certain timers only work on certain pins. B) Changing a timer will impact other functions like millis(), delay() etc depending on which one you're mucking with. C) Each timer has different abilities to divide and pre-scale by certain amounts. D) Making something that suits all these needs is extra complex. I will have to solve this problem for myself with at least 2 fans controlled by 2 separate pins. Seeing as your license is MIT, maybe I could contribute my changes back to your library and maybe make derivative classes which handle these specialties as I understand them more. I'm pouring through the actual datasheet at this point to better understand what they did in the hardware so I can understand the software implications. But I am kind of envisioning a derivation which might support certain frequencies - kinda "app specific" if you will.

GiorgioAresu, 2020-04-16:
If the registries work correctly I could just add a parameter to `setup()`. Please let me know if that works

steakknife, 2020-06-27:
[31 kHz is out-of-spec](https://www.handsontec.com/pdf_files/4_Wire_PWM_Spec.pdf) for the PWM fan standard, i.e., it's 25 kHz optimal and 21-28 kHz acceptable. Using 31 kHz will cause problems but it's impossible to know without testing every fan.

[Here's an (untested by me) method](https://arduino.stackexchange.com/a/25636) for setting PWM 25 kHz on Uno that looks reasonable.

sergeant82d, 2022-01-22:
I can confirm that using a timer to put out 25KHz does work with this library, when used as only a monitor, not for controlling the fan. I used the few lines I found online [here:](https://create.arduino.cc/projecthub/tylerpeppy/25-khz-4-pin-pwm-fan-control-with-arduino-uno-3005a1)

Thanks for some nice and useful code!