GiorgioAresu/FanController #13: Not sure this is a bug or just an artifact of the hardware I am using

haloway13, 2020-10-09:
Might be useful to someone else or maybe someone with greater knowledge can fill me in as to the behavior I am seeing.

Until I wrapped the input line with a >0 test the fan would temporarily try to get to the set speed but would then read a 0 and turn it off. Initial speed was 100%.

`int input = Serial.parseInt();
    if (input>0)
      {
...
}
}//End Main Loop`

haloway13, 2020-10-09:
Oh, I suppose you would like details on the hardware.... hah!

Mega 2560
With a fan from Bgears: https://www.amazon.com/gp/product/B07FHM5DJZ/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

Initial RPMs are correct at around 5300. Once I change the RPMs the reported RPMs jump into 15000 range for a 10% duty cycle. I can drop it as low as 4% duty cycle and the fan can't be spinning more than 300 rpm or so yet the serial monitor reports 15.4k

Love the library so far!

GiorgioAresu, 2020-10-10:
Hi, about the first comment, could you try setting the serial console you're using to "No line ending" or something similar? Regarding the second, does changing SENSOR_THRESHOLD value make any difference?
Thanks :)

haloway13, 2020-10-12:
"No line ending" did the trick for it not resetting the speed to 0.

Sensor threshhold did not seem to make much of a difference.

Thanks!
Tim