GiorgioAresu/FanController #15: How to stop the fans?

Gishanx, 2021-02-12:
I can't seem to find a way to stop the fans. The obvious setDutyCycle(0) doesn't work.

In forums I've read that you can turn it off by switching pin mode to OUTPUT and digitalWrite(pin, LOW).
But I don't know what effect that would have while this libraries code is running (problems along the line, how to start the fans again?)

I would highly appreciate if someone could implement a method stop() and maybe restart()

ufoDziner, 2023-07-01:
Just started to use this library.  It's great, but I can't stop the fans either.  Closest I can get is ~5% without side effects.  Any chance there is a fix in the works?  Thanks!

kenkit, 2024-05-22:
You can use another input and a mofset as mentioned in this other diagram.
You only need the mofset part 
https://github.com/sker65/esphome-fan-controller